makerbase-mks/MKS-Robin-Nano-V2.X #1: Support for MKS Robin Nano V2.0

HulusiAKAR, 2020-07-27:
There are many documents and videos for MKS robin nano v1.2 on the internet, but there are no documents and videos for robin nano v2.0. We have difficulty in installing. For mks robin nano v2.0 we want editable robin_nano35_cfg.txt. If I knew this would happen, I would get v1.2 instead of Mks robin nano v2.0.

makerbase-mks, 2020-07-28:
Hi! You can download https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware marlin for MKS Robin Nano V2, and MKS Robin Nano V2 need't config.txt file.

HulusiAKAR, 2020-07-28:
Thanks to the config.txt file, the settings were made very easily. In the new update, you should definitely edit with config.txt. If I knew I would buy the old version MKS Nano v1.2.

makerbase-mks, 2020-07-29:
MKS Robin Nano V2 firmware will add more parameters in the online adjustment function, parameter modification can be achieved without configuration files, so stay tuned

closed 2020-09-02 by makerbase-mks