makerbase-mks/MKS-Robin-Nano-V2.X #2: Help for Bltouch setup and settings for MKS Robin Nano v2.0

HulusiAKAR, 2020-07-27:
I could not adjust bltouch settings for MKS robin nano v2.0. Can you help me?

makerbase-mks, 2020-07-28:
Hi! Enable BLTOUCH function, you can refer to https://www.youtube.com/watch?v=CB1x841_WYo

HulusiAKAR, 2020-07-28:
I watched this video. I tried but I couldn't. There is no example for this video in MKS Nano v2. I would be glad if you could help.

makerbase-mks, 2020-07-29:
We will introduce how to use BLTOUCH and configuration method in MKS Robin Nano V2 Wiki.

HulusiAKAR, 2020-07-29:
When?

Tenteni, 2020-09-01:
In my experience, there is yet no way to introduce commands for the BLotuch in MKS Robin Nano V2. However, the BLTouch (3DTouch in my case) works very well with MKS 2.0 software, or with the generic Marlin 2.0.6.1. I You only have to enter the 'home Z' or the 'home all' commands, and the 3dtouch probe immediately deploys its sensor rod  and performs two successive measurements of the bed position. Of course, one has had previously to uncomment line 907 of configuration.h and compile the firmware again.

MOOKIYA, 2021-01-17:
I followed it the same way, but an error occurs.
"'servo_init' was not declared in this scope", "'probe' was not declared in this scope"
Can you help me?