makerbase-mks/MKS-Robin-Nano-V2.X #3: No graphics

Tenteni, 2020-08-25:
After a lot of work I have finally had a partial success building a Platformio project for my new MKS-Robin-Nano V2 board. It is installed now in a veteran Hictop printer, as a substitution for its original MPX.3,  8bits board, which was out of order.

From Marlin 2.0.6 standard software and edited config files, the compiler produced a ROBIN_NANO35.bin and an Assets folder, which are well accepted by the printer through a microSD card.

 As far as I can see, the configuration of the machine has been loaded well, and it changes the .bin extension to .CUR. However, the TS35 tft screen does not produce images at all, but some texts and a number of white squares instead of the asset images. As text is present in some screen elements, I have been able to check that the machine configuration has been correctly updated and that the menus work well.

I did follow MKS advice to introduce in the configuration.h file a new line reading #define TFT_LVGL_UI_SPI, but perhaps I have missed some other necessary change or addition in that respect. ¿Could you please help me to solve this problem?
Thanks in advance

themacboy, 2020-08-25:
Try updating with folders mks_pic and mks_font folder, you can't find it in
the firmware folder

makerbase-mks, 2020-08-26:
1. You can try to update again
2. You can download mks_font and mks_pic update it

Tenteni, 2020-08-26:
I followed your suggestion and it worked well, thank you !!!!!

Tenteni, 2020-08-26:
This last comment of mine was addressed to themacboy. I had not yet seen makerbase-mks's kind comment.

Regarding makerbase comment, I will expand a little on the reasons of my search:
My maker base-mks board came with a Marlin 2.5.3 firmware factory installed. That firmware did no offer in its menu the different configuration choices it was supposed to contain, and I thus could not configure my machine settings, unless I were able to customise the configuration.h file and compile it. That was what I did, and I chose Marlin 2.0.6 because it is announced to include full support for MKS Robin nano V2.

However, after compiling it and having solved the problem with the pictures, it still has a very limited menu options. For example: neither Bltouch nor Wifi options appear in the menu, nor do options for changing movement configurations and other facilities that MKS-Robin nano V1.2 has.

makerbase-mks, 2020-08-26:
@Tenteni 
Our Github now has the latest firmware, more parameters can be set through the TS35 screen, you can try it.
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware

Tenteni, 2020-08-26:
Thank you Makerbase-Mks!
I downloaded the firmware from the link provided. Before any editing, I just dropped into the microSD the robin_nano35.bin as it comes plus the pictures and fonts folders. It installed well and apparently provides more menu options, but pictures are  substitutes by a lot of white squares with the label 'no data'. I hope this will improve after editing the configuration files and building a new project.

makerbase-mks, 2020-08-29:
I’ll close this question for now.

closed 2020-08-29 by makerbase-mks

Tenteni, 2020-08-29:
We have repeatedly tested the GitHub Mks robin nano Marlin 2.0 bugfix, suggested by you 3 days ago, but it repeatedly failed to compile when TMC2209 was the specified driver. It does compile, however, with  TMC2209 standalone, but does not offer any TMC2209 configuration option in its menu.
I have also tested the new generic Marlin 2.0.6.1 and it works pretty well. It does recognise TMC2209 and offers its configuration options in the mms TFS35menu.
The main problem of this new generic Marlin configuration, appears to be that it does not recognise the mks-wifi. Neither is there a bltouch menu, although in fact it operates perfectly with the installed 3dtouch when homing.   
There remain also some minor problems, such as a low delay when introducing characters or commands through the TFT, what causes many repetitions.
When line 2225 in configuration.h "#define SPI_GRAPHICAL_TFT"is habilitated, a printer status menu is superposed over the graphical menu. That would be  a  useful menu if it were available on demand, but not always.