makerbase-mks/MKS-Robin-Nano-V2.X #4: The printer does not print

Tenteni, 2020-08-31:
In the 'no graphics' issue I described some problems I have encountered to set up a new Makerbase-mks-Robin-Nano v2.0 board with TMC 2209 drivers, in accordance with my old Hictop DP17 printer configuration.

I provisionally chose and configured Marlin 2.0.6.1 firmware, because it provides a reasonable menu, and  TMC2209 motor drivers are working in UART mode. The main disadvantage of this firmware is that it does not allow  axis offsets in the menu, and that MKS wifi is not recognised. I decided then to print a Gcode file from the SDcard: a new support for the 3Dtouch.

In the attached picture, the printing screen is shown. Apparently the Gcode file has been recognised and loaded, but no action is taken by the printer in response to the Gcode instructions.

[image]

Any suggestion, please?

makerbase-mks, 2020-09-02:
Gcode file name is long?

Tenteni, 2020-09-02:
It was 15 characters long when I first tried. I renamed with a 7 characters name and it does not print it from the microSD either. However, it does print the same gcode file from Cura through an USB connection, and it prints smoothly and silently.

Tenteni, 2020-09-10:
In my opinion, this MKS version of Marlin  bugfix  2.0.x has solved several old problems very well, notably the integration of TMC2209 in the menu, and the menu options to modify the probe-to-nozzle offsets. However, as far as I know there is still no way to print from the SDcard, nor does this version's menu offer in Settings/'Configuration'/'Adavance (sic) settings' any option to configure the WIFI connection. It is indeed a pity, because without SD and wifi printing, the only printing option left is the USB connection.
Please let me know if there is anything I can do to solve this two standing problems.

makerbase-mks, 2020-09-14:
@Tenteni Can you try this version?
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware

Tenteni, 2020-09-14:
From that new version you are proposing, the only part I'm able to test is the Robin_nano35.bin file in the Firmware folder. I'm unable to download the other files in their present format,  to build with them a customised new platformio project. That  version of the Robin_nano35.bin file from the Firmware folder performs very poorly, because it does not recognise the tmc2209driver, nor the wifi card, nor admits any useful customization.

mks-viva, 2020-11-03:
The latest version of the firmware already supports wifi file transfer, but there will be other bugs, our engineers are fixing

leofaure, 2021-03-06:
Hola Tenteni hace unos dias adquirí una mks robin nano v2.0 y tengo muchas dudas, creo que hablas español, podríamos ponernos en contacto para ayudarnos mutuamente y poder colaborar con este hardware en un idioma que domino.Gracias!

Tenteni, 2021-03-07:
Hola Leofaure, yo tengo, desde el año 2016 una impresora Hictop 3DP17. La he utilizado con asiduidad, pero para poder hacerlo he tenido que resolver problemas diversos a lo largo de estos 5 años. La ventaja es que he aprendido algunas cosas en ese proceso y además me he divertido.
Compré la placa madre Robin nano V2.0. con drives TMC2209  y tarjeta WiFi, en Julio del año pasado para substituir placa original de la Hictop, que tenía una avería. El proceso de adaptación fue complicado, porque el firmware de MKS no estaba todavía bien desarrollado. El que venía con la placa no me funcionaba y hasta Noviembre de 2020 no apareció en GitHub una buena versión de firmware para la Robin Nano V2.0, que -una vez compilada- fuera plenamente operativa en mi impresora. A final de enero de 2021 la substituí por otra versión de GitHub más actualizada, que ha mejorado en algunos detalles.
Como sabes, lo más delicado es hacer las modificaciones necesarias en el fichero configuration.h,  para los parámetros de cada impresora. Luego la compilación la hago con Visual Studio Code, que encuentro muy lioso pero que al final acaba funcionando.
Si tienes alguna duda en concreto estaré encantado de charlar contigo y de aportarte lo que buenamente pueda para que llegues a un buen entendimiento con tu nueva Robin nano V2.0.  En principio, convendría continuar por esta vía, por si algo de lo que escribamos sirve de ayuda a otros compañeros de lengua española.

leofaure, 2021-03-07:
Hola gracias Tenteni, pues al parecer estoy preparando correctamente mi build, con mis tmc2209 y doble stepper z, pero al hacer un m122 gode me da un error en E, haver si conseguimos algo en claro. este el el resultado del M122:
< 20:17:35: axis:pwm_scale/curr_scale/mech_load|flags|warncount
< 20:17:35: X	Y	Z	Z2	E
< 20:17:35: Address		0	0	0	0	0
< 20:17:35: Enabled		false	false	false	false	false
< 20:17:35: Set current	800	800	800	800	800
< 20:17:35: RMS current	795	795	795	795	1436
< 20:17:36: MAX current	1121	1121	1121	1121	2025
< 20:17:36: Run current	25/31	25/31	25/31	25/31	25/31
< 20:17:36: Hold current	12/31	12/31	12/31	12/31	12/31
< 20:17:36: CS actual	12/31	12/31	12/31	12/31	0/31
< 20:17:36: PWM scale					
< 20:17:36: vsense		1=.18	1=.18	1=.18	1=.18	0=.325
< 20:17:36: stealthChop	true	true	true	true	false
< 20:17:36: msteps		16	16	16	16	256
< 20:17:36: interp		true	true	true	true	false
< 20:17:36: tstep		max	max	max	max	0
< 20:17:36: PWM thresh.	98	98	658	658	3170
< 20:17:36: [mm/s]		100	100	3	3	30
< 20:17:36: OT prewarn	false	false	false	false	false
< 20:17:36: triggered
< 20:17:36:  OTP		false	false	false	false	false
< 20:17:36: pwm scale sum	14	14	14	14	0
< 20:17:36: pwm scale auto	0	0	0	0	0
< 20:17:36: pwm offset auto	36	36	36	36	0
< 20:17:36: pwm grad auto	14	14	14	14	0
< 20:17:36: off time	3	3	3	3	0
< 20:17:36: blank time	24	24	24	24	16
< 20:17:36: hysteresis
< 20:17:36:  -end		-1	-1	-1	-1	-3
< 20:17:36:  -start		1	1	1	1	1
< 20:17:36: Stallguard thrs	0	0	0	0	0
< 20:17:36: uStep count	24	8	8	24	0
< 20:17:36: DRVSTATUS	X	Y	Z	Z2	E
< 20:17:37: sg_result	0	0	0	0	0
< 20:17:37: stst						*
< 20:17:37: olb						
< 20:17:37: ola						
< 20:17:37: s2gb						
< 20:17:37: s2ga						
< 20:17:37: otpw						
< 20:17:37: ot						
< 20:17:37: 157C						
< 20:17:37: 150C						
< 20:17:37: 143C						
< 20:17:37: 120C						
< 20:17:37: s2vsa						
< 20:17:37: s2vsb						
< 20:17:37: Driver registers:
< 20:17:37: X	0xC0:0C:00:00
< 20:17:37: Y	0xC0:0C:00:00
< 20:17:37: Z	0xC0:0C:00:00
< 20:17:37: Z2	0xC0:0C:00:00
< 20:17:37: E	0x00:00:00:00	 Bad response!
< 20:17:37: Testing X connection... OK
< 20:17:37: Testing Y connection... OK
< 20:17:37: Testing Z connection... OK
< 20:17:37: Testing Z2 connection... OK
< 20:17:37: Testing E connection... Error: All LOW

gracias por todo!

leofaure, 2021-03-07:
también me gustaría saber si hay manera en el display mas ts35 v2.0 si hay manera de que aparezca correctamente conoofiguraciones de la impresora: tipo de impresora, tamaño, inversión de motores y endstop, y todo lo que aparece en la versión mas robín nano 1.
gracias!!

Tenteni, 2021-03-08:
Para poder contestar a tu comentario sobre la configuración del TMC2209, he intentado obtener en mi máquina el resultado del comando directo M122. En primer lugar he probado desde el Cura -por la conexión Wifi- pero ocurre que en el Cura yo no he sabido encontrar una forma de ver el texto del resultado de los comandos directos, ya que en la ventana ‘Monitor’ no hay sitio para que aparezca el volcado correspondiente.

Luego he intentado enviar el M122 con Pronterface. Pero el Pronterface no reconoce la wifi y tampoco he podido hacer funcionar la conexión Usb, que usaba yo antes de conseguir habilitar la Wifi de la impresora. El problema debe estar en la asignación de puertos serie que hay al principio de ls sección ‘machine’ del configuration.h.

En el build que tenía habilitado en Noviembre, yo definía en primer lugar #define SERIAL_PORT 3 y a continuación #define SERIAL_PORT_2 -1 . El ‘configuration’ compilaba bien y el usb funcionaba sin problemas.

Sin embargo, en la versión de enero, la instrucción #define SERIAL_PORT_2 -1, me daba error al compilar y me vi obligado a comentarla, con lo cual ahora veo que perdí la conexión usb. Intentaré arreglarlo en cuanto pueda.

En cuanto al driver del extrusor, tengo un problema pequeño pero extraño. En el configuration.h yo he habilitado solamente el E0, y con el driver tmc2209. Sin embargo, en las pantallas de configuración de motores me aparecen tanto el E0 como el E1. El E0 con un límite de corriente de 800mA y el E1 con un límite de corriente de -1mA. No obstante, en las pantallas Extrusion y Filaments (de Tools) sólo aparece el E1. Supongo que será un error de los Assets que carga la pantalla al iniciarse.

Respecto a a cuestión de si es posible que aparezcan en el TS35 v2.0 los datos de tipo de impresora, tamaño, inversión de motores, etc. etc. Sólo te puedo decir que en mi máquina no aparece ninguno de esos datos. Supongo que no aparecen porque la Robin nano V2.0 no tiene las mismas capacidades de configuración ‘en caliente’ que tenía la placa Robin nano V 1.2
Saludos