makerbase-mks/MKS-Robin-Nano-V2.X #5: Nano V2 Wifi Module

WashingtonJunior, 2020-09-08:
When will Wifi function be available for Nano v2? Why we can't use the same firmware as Nano 1.2 (that supports wifi)? Thanks!

HulusiAKAR, 2020-09-10:
It has been 2 months since I bought the device and we are still waiting for an update. Hear our voices now.

makerbase-mks, 2020-09-14:
Thank you for your attention, we are developing the firmware that supports Wifi. Because Nano v1.2 have not compatible marlin 2.0.x, but, we will add wifi function for Nano V2 board with marlin 2.0.x

WashingtonJunior, 2020-09-20:
By following the tip available in https://github.com/makerbase-mks/MKS-Robin-Nano-V2/wiki/MKS_Robin_WIFI, I was able to turn on wi-fi in both Nano 1.2 and Nano v2 and the same issue also happens to both of them. When uploading a file, it gets stuck in "Uploading...". Any idea when this feature will be available? Thanks!

themacboy, 2020-09-20:
read below:

Test Wi-Fi

   -

   Connect Wi-Fi IP by Web
   -

   *This firmware version does not currently support file transfer*. We
   will add the function of transferring gcode file by WIfi In the future

Missatge de Washington C. Corrêa Jr. <notifications@github.com> del dia
dg., 20 de set. 2020 a les 17:02:

themacboy, 2020-09-20:
btw, where is the wifi icon?

WashingtonJunior, 2020-09-21:
> Test Wi-Fi - Connect Wi-Fi IP by Web - *This firmware version does not currently support file transfer*. We will add the function of transferring gcode file by WIfi In the future

Yeah, I saw that, that's why I ask about "when will it be available", since Bluer's original firmware is Marlin based and has this function properly. Everything would be easier if they release their source-code for Nano 2.x firmware, which works... 😞

tomcasa, 2020-10-17:
Hello , I am also expecting the support of the ESP8266 WIFI module. 
the MksWifi.bin is not loaded at startup.
Thanks a lot for your understanding

themacboy, 2020-10-19:
Please check Robin Nano wiki to know how activate WiFi module

tomcasa, 2020-10-19:
Hello, thanks for your help, I have updated the firmware to the last version (Mks-Robin-Nano-Marlin2.0-Firmware)

[image]

To selector button is running fine BUT neither
- the MksWifi.bin is not loaded at startup.
- the config file robin_nano35_cfg.txt is not flagger as 'CUR' 

[image]

Could you update the firmware about this problem ? 
I need this before end of October 
Thanks for your support and very good expertise.

WashingtonJunior, 2020-10-19:
Hi, @tomcasa 

You're talking about the original firmware that MKS uses on Nano Robin, that one uses the text file for settings. But for this Marlin version it won't work this way, you can set the configuration by recompiling the firmware. And you can enable the Wifi setting through this guide: https://github.com/makerbase-mks/MKS-Robin-Nano-V2/wiki/MKS_Robin_WIFI ... But, it's effortless, since it is not capable of sending files yet.
My original Two Trees Bluer firmware (with Nano Robin) works as expected and has a text configuration files like yours. What is your printer?

Cheers!

tomcasa, 2020-10-21:
> Hi, @tomcasa
>
> You're talking about the original firmware that MKS uses on Nano Robin, that one uses the text file for settings. But for this Marlin version it won't work this way, you can set the [...]

Thanks for your response, 
I have a flying bear printer, the MKS Gen_L V1 broke, so I want to upgrade it. 
https://flyingbear3dprinter.com/3d-printer

WashingtonJunior, 2020-10-21:
Right, and did you get a Robin Nano v2.0 board? If so, I think for now you won't have wi-fi yet. For Nano 1.2 board you could you use Bluer's firmware for that.

mks-viva, 2020-10-22:
MKS Robin Nano V2 has supported tarnsfer files, the file size <100m is OK and file size more than 100M may be fail, we are fixing it

themacboy, 2020-10-22:
How you can transfer files? Only cura plugin?

mks-viva, 2020-10-28:
The latest version firmware, supported transfer gcode files to TF card via wifi

themacboy, 2020-10-28:
How?

Missatge de mks-viva <notifications@github.com> del dia dc., 28 d’oct. 2020
a les 9:49:

tomcasa, 2020-10-28:
> ersion firmware, supported transfe

Thanks @MKS-Viva and @MKS-Sean 

After update some images assets are missing. 

[image]
[image]

mks-viva, 2020-10-29:
@tomcasa Hi! You need copy assets folder to TF card update it and the path is .pio->build->mks_robin_nano35

themacboy, 2020-10-29:
to refresh all libraries and assets folder with the current source you only
need to remove the "libdeps" folder on the ".pio" folder.

Then when compiling you will download last versions.

Missatge de mks-viva <notifications@github.com> del dia dj., 29 d’oct. 2020
a les 2:24:

mks-viva, 2020-10-29:
@themacboy Can it work?

tomcasa, 2020-10-31:
@mks-viva, Thanks, with your advice I arrived here 👍 

[image]

add I found the miising asset new pictures files in this folder : 

[image]

tomcasa, 2020-10-31:
The next issue is when using the MKS cloud APK : 

I can't load the Udisk or SDcard root folder 
and I can't upload the gcode file

The loading is infinite on this page : 

[image]

tomcasa, 2020-10-31:
And from CURA + MKS wifi plugin 
still not found :  
[image]

mks-viva, 2020-11-03:
@tomcasa 
Can you check wifi on TS35 display screen?

tomcasa, 2020-11-03:
Hello @mks-viva 
The WIFI config on TS35 works very good. 
The board is connected and the webpage green works also. 
Do you need a screenshot of this screen ?

themacboy, 2020-11-04:
Hello, can you publish a video explaining how to upload any gcode using web
interface? Thx ina advance !

Missatge de tomcasa <notifications@github.com> del dia dt., 3 de nov. 2020
a les 14:00:

tomcasa, 2020-11-04:
As previously said, the gcode UPLOAD doesn't work neither via the MKS cloud APK or CURA plugin.

themacboy, 2020-11-04:
> 
> 
> As previously said, the gcode UPLOAD doesn't work neither via the MKS cloud APK or CURA plugin.

Hello @tomcasa my request is not for you.

I hope that @mks-viva  @makerbase-mks can provide us recording a few videos, with file transfer configurations and a few examples.

WashingtonJunior, 2020-11-06:
It's already working, just tested it yesterday. It reaches 140Kbytes/s here with me and the uploaded file works perfectly now.

mks-viva, 2020-11-11:
> > As previously said, the gcode UPLOAD doesn't work neither via the MKS cloud APK or CURA plugin.
> 
> Hello @tomcasa my request is not for you.
> 
> I hope that @mks-viva @makerbase-mks can provide us recording a few videos, with file transfer configurations and a few examples.

Nano V2 Wiki about how to use MKS Robin Wifi: https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/MKS_Robin_WIFI

themacboy, 2020-11-12:
Is there a option (any) to upload gcode files using the web interface?

WashingtonJunior, 2020-11-12:
> Is there a option (any) to upload gcode files using the web interface?

As far as I know, not yet. I've been using MKS Wifi Plugin for Cura for that. But, it looks like there a bug as the plugin doesn't seem to know when the printing is finished (therefore not allowing a new upload, unless you close and open Cura again). I believe it's a firmware issue, as the original firmware (for TwoTrees Bluer) doesn't have this behaviour.

mks-viva, 2020-11-13:
> Is there a option (any) to upload gcode files using the web interface?

Of course! But you need use wifi's AP mode.

themacboy, 2020-11-13:
Sorry, I'm not skilled. Not sure how to do all that.

tomcasa, 2020-11-13:
I can't understand why we need to use the AP mode as the goal is to connect the printer as a STAtion to the AP of the domestic WIFI router. 
Thanks for your support.

mks-viva, 2020-11-14:
You connect wifi by ip and use web, you can upload gcode file.

mmillonida, 2020-11-15:
I have everything setup on the Robin Nano V2 board, but when i try to upload from Cura, the printer says communication error and I can't connect via the web either. I can't even print from my SD card as it shows me the files but just freezes on the file screen and I need to restart the printer just to get back to the menu. PLEASE HELP!!! I really like the functionality of the Robin Nano and TS35 screen but can't print!!!

WashingtonJunior, 2020-11-15:
> I have everything setup on the Robin Nano V2 board, but when i try to upload from Cura, the printer says communication error and I can't connect via the web either. I can't even print from my SD [...]

Does the printer shows that it's connected to wi-fi? And if you type its IP on a browser in the same network does it shows a green page? If not, check your wi-fi settings (SSID and password) and check if it's in STA mode. Cura plugin will only work if at least this page shows up.
About the freezing, do you have previous files on the SD card? Mine was freezing and it was old files (with corrupted thumbnails from previous MKS Wifi plugin) that was causing it. I've been using on a Nano v1.2 with a TFT35, but a friend of mine has your current setup (Nano V2 and TS35) and it's working either.

mks-viva, 2020-11-30:
MKS Robin Nano V2+MKS TS35+MKS Robin WIFI, at this stage, can be use it normally. So, I will close this issues.

closed 2020-11-30 by mks-viva