makerbase-mks/MKS-Robin-Nano-V2.X #6: Nano V2 Wifi Module Not Working

HulusiAKAR, 2020-09-10:
It has been 2 months since I bought the device and we are still waiting for an update. 
When will the device receive a Wifi update? We cannot use the wifi plug.
I have an error every time. We wouldn't buy this device if I knew that would happen.
Hear our voices now.

makerbase-mks, 2020-09-14:
Thank you for your attention, we are developing the firmware that supports Wifi.

mks-viva, 2020-10-22:
@HulusiAKAR MKS Robin Nano V2 has supported tarnsfer files, the file size <100m is OK and file size more than 100M may be fail, we are fixing it

themacboy, 2020-10-22:
How you can transfer files to SD? Only cura plugin?

mks-viva, 2020-10-28:
The latest version firmware, supported transfer gcode files to TF card via wifi
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware

mks-viva, 2020-11-30:
MKS Robin Nano V2+MKS TS35+MKS Robin WIFI, at this stage, can be use it normally. So, I will close this issues.

closed 2020-11-30 by mks-viva