makerbase-mks/MKS-Robin-Nano-V2.X #8: Nano V2 on Delta, is it possible?

pistachin2, 2020-09-22:
Hello everyone, I bought this board for the good results that V1.2 has given me.
The problem is that I have it mounted on a Delta and I have no way to make it work.
One of the problems is that delta use max endstop, and this board does not have them, but it is indicated that it is functional with Delta.
The movements were performed well, when moving the z axis all the motors work to join or lower the car.
But when I do a homing, it goes crazy and the engines give a terrible blow.
Does anyone have this plate mounted on a delta?
Thank you very much in advance.

mks-viva, 2020-10-22:
I think you need check your manchine limit NC or NO status

ikkez, 2020-11-02:
I have successfully mounted this board on a delta last weekend. I also use max endstops and had no problems with it. On the board, the X-STOP and Y-Stop ports are max-endstops, even if it's not written there. Use Z+ Port for the Z endstop and Z- for a probe.

mks-viva, 2020-11-03:
@ikkez 
Hi! Can you share your onfig to others, thank you!

ikkez, 2020-11-03:
@mks-viva my current configuration https://github.com/ikkez/Mks-Robin-Nano-Marlin2.0-Firmware/commit/c139ef584d994a6d506c6f4806bee9247e7b21a4
3d touch, wifi, leveling works. Some buttons on the FTF are not doing what they should yet, but so far I can print now.