makerbase-mks/MKS-Robin-Nano-V2.X #10: Auto leveling stop working after couple of prints

helgjohn, 2020-10-03:
Have an issue with 3d touch sensor it dosent seem to work after a short while It dosent mater if its mks or skr board same issue on both brands. The probe does its thing and collect all data about the bed and so on but does not use the collected data in order to adjust the nozzle. So the ting i have now is a advanced end stop. have disabled and enabled every ting i can think of that correspond to the sensor. And the only differ is some times it crash in to the bed. The only remedy i have had any suksess with is re installing marlin and then it works for a short while. So now i am lost for options. The video for installment  by MKS is a bit messy but i have to the best of my ability followed the steps, there is no video for MKS SGenl v2 (1769 chip) so i guess the instructions are the same the same as Sgenl (1768 )

mks-viva, 2020-10-22:
@helgjohn Use MKS SGEN_L V2 motherboard?

helgjohn, 2020-10-22:
> @helgjohn Use MKS SGEN_L V2 motherboard?  
Yes MKS SGENL V2 is now working but haven`t done many prints so the issue can lurk still. in order to get it up and running new install was required i now run at marlin ver 2.0.7.. 
-Have also tested with new board / sensor and Wire ring and connectors it make  NO DIFFER. 
-Have also tested all settings one by one also in diff combo and the only thing that works in the end is new install whit simplest set up.. 
-If i activate Dual X on separate drivers and auto align it works fine and some times it doesen`t then the homing also is whanky so i conclude there is an issue with software and board Sensor combo. But no luck pinpointing it yet.
" there is no video for MKS SGEN L v2 (1769 chip) so i guess the instructions are the same the same as Sgenl (1768 )"
 "Have an issue with 3d touch sensor it dosent seem to work after a short while It dosent mater if its mks or skr board same issue on both brands. "