makerbase-mks/MKS-Robin-Nano-V2.X #11: Add the axis info

DvoraNoob, 2020-10-20:
In the option to move does not have the information of the axis, for example: X:110 Y:50 Z:4, so i have no way of knowing the position of the axis

mks-viva, 2020-10-22:
@DvoraNoob 
Hi! We are considering adding this feature

duvan60, 2021-01-08:
He instalado ROBIN NANO V2.0 en una creality CR10 y no actualiza para poder programar área de impresión en la pantalla y poder escoger el tipo de maquina como dicen en los videos de youtube. Puedes ayudarme con la actualización correcta o que pasa con la actualización en la tarjeta SD.

mks-viva, 2021-01-08:
@duvan60 
Hi! You can download firmware bin and copy Robin_nano35.bin to tf card, test it. 
[image]

duvan60, 2021-01-08:
> @ duvan60
> ¡Hola! Puede descargar el firmware bin y copiar Robin_nano35.bin a la tarjeta tf, probarlo.
> [image]

hola @mks-viva, ese es el firmware que tengo descargado en la impresora y es por eso que me pareció extraño, ya que MAKERBASE nos dice que debía actualizar y nuevas funciones vendrían, pero en mi electrónica no ha sucedido nada. sera posible que sea necesario una TFT35 y que no funcione en la TS35¿ les agradezco mucho la ayuda.

mks-viva, 2021-01-09:
You used Robin Nano V2 motherboard, what other screens are there?

closed 2021-04-15 by DvoraNoob