makerbase-mks/MKS-Robin-Nano-V2.X #12: Update firmware and Printing from SD alway hang

billydragon, 2020-10-22:
I have MKS Robin Nano V2 + TS35 board. Firmware update by MKS - Marlin for Robin V2. Update from SD card very hard. If lucky I can update at first time, normaly need to format clean SD card, copy FW file to SD card then try update. if not lucky, need to do again format SD, copy FW file ....
After finish update and config every things done. Printer work fine. Printing from PC via USB cable have no problem, 4-5 hours done. But when i try print from SD card randomly hang. No one time with success. I think reading and manager SD card have problem. Try all 3-4 card i have with the same result.

RGDS.

mks-viva, 2020-10-23:
What is the SD card format you set? FAT32 or other and your tf card size?

billydragon, 2020-10-23:
Card is alway formatted Fat32.

billydragon, 2020-10-23:
I used 2 Card 16gb and 32gb Samsung brand name

AdamMulica, 2021-11-05:
Well, you can only wait for your SD slot to collapse completely. I also had a problem with the SD port 2 times. And it hasn't been working for a couple of days.

https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/issues/42