makerbase-mks/MKS-Robin-Nano-V2.X #13: how can I use TS35 on klipper?

system3rd, 2020-10-29:
hello I'm trying to use klipper on mks robin nano V2.0
mostly working fine except ts35 screen.
is there any possibility to use TS35 on klipper?
anyone who can success to use it?

mks-viva, 2020-11-03:
Hi! Can you provide relevant tutorials? We will also explore how to support TS35 display...

Loreagle, 2020-11-04:
> Hi! Can you provide relevant tutorials? We will also explore how to support TS35 display...

This is the link of the firmware, at the moment is impossible to use the TS35 on Klipper but it would be awesome if you can give us the opportunity since Klipper is a very powerful firmware and a lot of people prefer it over Marlin.

https://github.com/KevinOConnor/klipper

system3rd, 2020-11-13:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/290

bigtreetech's tft35 seems to working but not perfect yet. character generator support needs for klipper.

mks-viva, 2020-11-14:
@system3rd 
Hi! I want to kown, Have you try to use LCD12864 on MKS Robin Nano V2 motherboard and run Klipper firmware?

mks-viva, 2021-01-20:
@system3rd Hi! I has pull requests for klipper, https://github.com/KevinOConnor/klipper/pull/3797

mks-viva, 2021-01-20:
So, i will close this issues temporarily.

closed 2021-01-20 by mks-viva

Loreagle, 2021-01-20:
> So, i will close this issues temporarily.

yes but what about TFT35 on MKS Robin Nano 1.2, there is no way to support it?

system3rd, 2021-02-05:
sorry for late reply. I tried it but TS35 is not working with this way. (KevinOConnor/klipper#3797)

mks-viva, 2021-02-06:
@system3rd 
Have not yet supported.