makerbase-mks/MKS-Robin-Nano-V2.X #14: CAN'T use UART TMC2209 stepper driver

tomcasa, 2020-11-18:
After having placed the jumper as described 

[image]

the TMC current setting is not present in the MKS TS35 screen. 
Does that mean the firmware must be recompiled by the final user ? 

[image]

Thanks for your support.

mks-viva, 2020-11-19:
Hi! You need define drivers type in configuration.h file and rebuild firmware.

mks-viva, 2020-11-30:
Has this problem been solved? If there is no response, then I will temporarily close this question.

closed 2020-11-30 by mks-viva