makerbase-mks/MKS-Robin-Nano-V2.X #17: Cant use PWC and UPS module. Zrise don't work

misiek271, 2020-11-23:
Hi i have 
MKS RObin Nano v.2 and TS35
How need i connect pwc and ups modules. To work: auto off function ,power loss recovery and z rise 
now i have got connectet PWC to  PW-det pin and PW- off pin. printer turn onn and of.
UPS is only connected to + and - to charge and discharge capacitors.
But when i take out power AC my z axis wont rise. 
i attach my cfg files. maybe i make some mistake?

mks-viva, 2020-11-23:
Hi! You can refer to MKS Robin Nano V2 Wiki about how to use MKS PWC and MKS UPS on Nano V2.

misiek271, 2020-11-23:
Yes i know tha. But in wiki ther is show that ups and pwc are connected to the same pw_det pin. So i presume that is mistake

mks-viva, 2020-11-23:
Haha! It seems that I need to do a tutorial about how to use MKS PWC and MKS UPS on MKS Robin Nano V2 motherboard at the same time

misiek271, 2020-11-23:
That will be great:).

mks-viva, 2020-11-24:
Hi! I find it and we can define #define POWER_LOSS_PIN to other pin, for example, #define POWER_LOSS_PIN  PA4 so all problems can be solved.
[image]

misiek271, 2020-11-24:
Yes i realize that. But i measure pwc and ups. And pwc is faster. I mean when you disconnect ac plug pwc turn off relay. About 1s after you unplug ac. And ups is based on voltage comparator so it make logical high when voltage level drop below some point. Ergo: when you disconnect ac plug it take few second to make dc  12v drop let say to 10v and then ups module give us logical high. So you have lose 2v for waiting to dropout voltage and this 2v is drained from powwer supply and capacitor so you lose. About 20%energy which should be used for closing and rise z axis. So in my case when i have fans and leds connected to 12v not through the mks. Its drain capacitors faster then ender or other printer and my z axis lift up only 4mm. So for my is beter to use pwc as outage detector but to do this you must disable kill pin in pins. H :) btw you should make ups that is conected to ac only for measure adc of 220v and it should set logic high level based on ac check and not by power supply dropout voltage coz its lose precious mA for waiting for dropout voltage:)

mks-viva, 2020-11-25:
Haha! MKS UPS module is directly detects the power failure of the external AC power supply, and does not use the DC voltage comparator method. Therefore, the detection method of the MKS UPS will respond faster, and more power will be used to Z lift move.

This is MKS UPS detects circuit:
[image]

misiek271, 2020-11-25:
Sorry.. I use this module from btt with capacitors.https://a.aliexpress.com/_BTGBFb
Btw. Your boards are better then btt and cheaper. You hav gotbetter tutorial datasheet and help on github

mks-viva, 2020-11-26:
Haha! This MKS UPS module: https://www.aliexpress.com/item/4000910659196.html?spm=a219c.12010612.8148356.3.634571d5fUYxJJ
If you know there is an MKS UPS module, I believe you will choose it

misiek271, 2020-11-26:
Sure in nex printer. Coz i compare your product to others company and its better. Thanks for help this topic can be close mks my first problem was solved. Fast and acurate that s what i like:)

closed 2020-11-26 by misiek271