makerbase-mks/MKS-Robin-Nano-V2.X #18: cant use filament detector with ts35

demeiji, 2020-11-23:
Hi. by following the steps of the MKS_FES wiki: Enable #define FILAMENT_RUNOUT_SENSOR, 
Enable #define NOZZLE_PARK_FEATURE, Add #define FIL_RUNOUT_PIN PA4, Enable #define ADVANCED_PAUSE_FEATURE, when compiling marlin I get an error ADVANCED_PAUSE_FEATURE currently requires an LCD controller. ADVANCED_PAUSE_FEATURE needs to be defined for the #define FILAMENT_RUNOUT_SCRIPT "M600"
Without enabling anything and simply connecting the filament detector to MT_DET1, my printer with mks pwc when the filament is finished turns off. turning it on again and activating resume starts printing from the beginning, not where it ended when it was turned off.
Is there any solution to make M600 work on ts35?
In my current build, without enabling any of the above, to properly resume printing from where it was shut down, do I have to enable anything?
thanks for your help

mks-viva, 2020-11-24:
Can you provide your firmware to me?

misiek271, 2020-11-24:
yes same problem to me. change runout script to m25. and where i click pause printing stops. but extruder dont unload filament. when i click resume extruder unload filament so something is wrong with that scoript coz it should unload first not after i click reasume.

mks-viva, 2020-11-25:
@misiek271 
Hi! Can you provide your vodies to me when you use change runout opation?

misiek271, 2020-11-25:
Ichecked the same goes with pause option. Heres video

misiek271, 2020-11-25:
And i attach next video first atempt extruder unload filament as before. Second atempt extruder dont unload filament but its not printing after reasume hotend moves but extruder make only linear advance move so gear move back and front and its not printing anything. I make black line on gear so it can be visible how its move. And if you can se when runout sensor dont have filament screen is black fo a while like reseting itself.its something wrong with pause function. My filament change setup in screen load and unload are 0. https://drive.google.com/folderview?id=1AMdf7lnrnXHmyWKkkqzWjOtDjpUtSiR5

misiek271, 2020-11-25:
[image]

misiek271, 2020-11-25:
[image]

demeiji, 2020-11-26:
Hello mks-viva

I have used the marlin from their github

With configuration 1 following the wiki MKS_FES gives the following error
#error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."

With configuration 2 (configuration 1 disabled) when finishing the filament the machine turns off and when turning it on and resuming it starts printing from the beginning, not from where it was turned off.

I have an MKS_PWC v2 installed
and it is configured following the wiki.
The printer turns on and off well although I will have another query with this topic later.

regards

[configurationh(1).docx](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5601549/configurationh.1.docx)
[configuration_ advh(1).docx](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5601550/configuration_.advh.1.docx)
[pins(1).docx](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5601551/pins.1.docx)
[configurationh(2).docx](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5601694/configurationh.2.docx)
[configuration_ advh(2).docx](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5601695/configuration_.advh.2.docx)
[pins(2).docx](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5601696/pins.2.docx)

mks-viva, 2020-11-26:
@misiek271 
Hi! We are testing and verifying the problem in your video.

mks-viva, 2020-11-26:
> Hello mks-viva
>
> I have used the marlin from their github
>
> With configuration 1 following the wiki MKS_FES gives the following error
> #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller [...]

About:  #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
You need enable LCD, for example, #define MKS_TS35_V2_0 display screen

demeiji, 2020-11-26:
Hello, sorry for my ignorance. Where is that line in configuration.h?
Or do I just have to add it in the LCD / controller selection (grafical LCDs)?
In the graphical TFTs section is already defined #define TFT_LVGL_UI_SPI // Robin nano v2.0 uses SPI
Regards

UrknallBigBang, 2020-11-30:
> Hi. by following the steps of the MKS_FES wiki: Enable #define FILAMENT_RUNOUT_SENSOR,
> Enable #define NOZZLE_PARK_FEATURE, Add #define FIL_RUNOUT_PIN PA4, Enable #define ADVANCED_PAUSE_FEATURE, when [...]

Change the Line in Marlin/src/feature/pause.h   #if HAS_LCD_MENU into #if HAS_LCD_MENU || HAS_FTF_LVGL_UI  and try compile again

misiek271, 2020-11-30:
Nope dont work. sanity check and error about emergency pareser or lcd

mks-L, 2020-12-09:
#define MT_DET_1_PIN                        PA4   // LVGL UI FILAMENT RUNOUT1 PIN
  #define MT_DET_2_PIN                        PE6   // LVGL UI FILAMENT RUNOUT2 PIN
  #define MT_DET_PIN_INVERTING                false // LVGL UI filament RUNOUT PIN STATE
Hi,If you use the LVGL UI, then it has its own fault detection function, just need to connect the device.No need to Enable Enable #define FILAMENT_RUNOUT_SENSOR.

demeiji, 2020-12-09:
> #define MT_DET_1_PIN PA4 // LVGL UI FILAMENT RUNOUT1 PIN
> #define MT_DET_2_PIN PE6 // LVGL UI FILAMENT RUNOUT2 PIN
> #define MT_DET_PIN_INVERTING false // LVGL UI filament RUNOUT PIN STATE
> Hola, si usa la propia IU de LVGL, entonces tiene su propia detección de UI LVGL función, solo necesita conectar el dispositivo. No es necesario habilitar Habilitar #define FILAMENT_RUNOUT_SENSOR.

Thanks for the reply. This is how I have the current compilation. I have a mks pwc installed and the problem is that when the filament runs out, the printer turns off and when I turn it on again and press continue printing, it starts from the beginning and not from where it was when the filament finished. I don't know if this behavior is correct or if when the filament is finished, the printer should simply be paused in order to change the filament and it should not turn off.

mks-L, 2020-12-09:
You can check to see if the wrong shutdown interface is connected to the fault detection interface.The MT DET port is used for fault detection function.The PW DET port is used for shutdown functions。

demeiji, 2020-12-09:
> You can check to see if the wrong shutdown interface is connected to the fault detection interface.The MT DET port is used for fault detection function.The PW DET port is used for shutdown functions。

misiek271, 2020-12-09:
> #define MT_DET_1_PIN PA4 // LVGL UI FILAMENT RUNOUT1 PIN
> #define MT_DET_2_PIN PE6 // LVGL UI FILAMENT RUNOUT2 PIN
> #define MT_DET_PIN_INVERTING false // LVGL UI filament RUNOUT PIN STATE
> Hi,If you use the LVGL UI, then it has its own fault detection functi

I check this solution . But problem didn t solved. when YOu out of filament. Printer stops. but when you click reasume. extruder unload filament and then start to print. I even changed in ts35 filament setings load and unload amount for 0 but this didnt fix it.

demeiji, 2020-12-09:
> > You can check to see if the wrong shutdown interface is connected to the fault detection interface.The MT DET port is used for fault detection function.The PW DET port is used for shutdown functions。

Thanks for your attention

Tthis is my configuration

mks pwc v2 connected in pw_det (5) and pw_of in (8). FES connected in M_det 1 (6)

In Configuration.h ：

//#define FILAMENT_RUNOUT_SENSOR

#define PSU_CONTROL
// # define PSU_NAME "Power Supply"

#if ENABLED (PSU_CONTROL)
  #define PSU_ACTIVE_STATE HIGH // Set 'LOW' for ATX, 'HIGH' for X-Box

in pins_MKS_ROBIN_NANO_V2.h
#define MT_DET_1_PIN PA4 // LVGL UI FILAMENT RUNOUT1 PIN
#define MT_DET_2_PIN PE6 // LVGL UI FILAMENT RUNOUT2 PIN
#define MT_DET_PIN_INVERTING false // LVGL UI filament RUNOUT PIN STATE

#define PS_ON_PIN PB2 // PW_OFF
#define KILL_PIN PA2 // PW_DET
#define KILL_PIN_STATE true //

When the filament is finished, pwc turns off the printer, regardless of whether M81 is added in the endscript or not when laminating the piece. Regardless of whether the switch is left automatic or manual from ts35

Regards

misiek271, 2020-12-09:
and What about unloading filament after easume print have solution for this? coz in my cas pwc works good. but reasume print or filament runout works wrong coz of unloading filament after reasume.

demeiji, 2020-12-09:
> y ¿Qué pasa con la descarga de filamento después de que la impresión de easume tenga una solución para esto? porque en mi cas pwc funciona bien. pero reanudar la impresión o el agotamiento del filamento funciona mal porque se descarga el filamento después de la reanudación.

Hi
You can tell me how you have connected and configured the MKS PWC in configuration.h and pins_MKS_ROBIN_NANO_V2.h?

misiek271, 2020-12-09:
#if HAS_TFT_LVGL_UI
  #define MKSPWC
  #ifdef MKSPWC
    #define SUICIDE_PIN                       PB2   // Enable MKSPWC SUICIDE PIN
    #define SUICIDE_PIN_INVERTING             false // Enable MKSPWC PIN STATE
    #define KILL_PIN                          -1//   // Enable MKSPWC DET PIN
    #define KILL_PIN_STATE                    true  // Enable MKSPWC PIN STATE
  #endif

  #define MT_DET_1_PIN                        PA4   // LVGL UI FILAMENT RUNOUT1 PIN
  //#define MT_DET_2_PIN                        PE6   // LVGL UI FILAMENT RUNOUT2 PIN
  #define FIL_RUNOUT_PIN PA4
  #define MT_DET_PIN_INVERTING                false // LVGL UI filament RUNOUT PIN STATE
 

[Configuration.txt](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5667832/Configuration.txt)
But i use mkspwc + mks filament sensor+ power off detection from (btt)
so i disable kill pin .

misiek271, 2020-12-18:
#mks-viva
Any news About Bad working filament script and unloading filament after reasume?

mks-viva, 2020-12-19:
I am sorry, I seem to forgot to inform you, I have updated the Wiki about the use of MKS FES on the MKS UI (MKS TS35)

misiek271, 2020-12-19:
No. Its not working. After reasume filament still unloading.

misiek271, 2020-12-19:
And its not problem with fes its problem with your pause script. when i pause print and then reasume it. extruder start unloading filament and then prints.

mks-viva, 2020-12-21:
> And its not problem with fes its problem with your pause script. when i pause print and then reasume it. extruder start unloading filament and then prints.

Ok, let's continue to test it

zonalimitatore, 2021-05-22:
Any solution? I ve a client that wanna 2 fil runout,because have 2 extruders.
The problem it's while compiling with this screen and mks robin v2

`Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp: In function 'void btn_ok_event_cb(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:169:7: error: 'pause_menu_response' was not declared in this scope
       pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
       ^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:169:7: note: suggested alternative: 'pause_menu_disp'
       pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
       ^~~~~~~~~~~~~~~~~~~
       pause_menu_disp
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:169:29: error: 'PAUSE_RESPONSE_EXTRUDE_MORE' was not declared in this scope
       pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/boards.h:24:0,
                 from Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:27:
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp: In function 'void btn_cancel_event_cb(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:228:35: error: 'pause_menu_response' was not declared in this scope
     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
                                   ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:228:5: note: in expansion of macro 'TERN_'
     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
     ^~~~~
Compiling .pio\build\mks_robin_nano35\src\src\lcd\extui\lib\mks_ui\draw_homing_sensitivity_settings.cpp.o
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:228:35: note: suggested alternative: 'pause_menu_disp'
     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
                                   ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:228:5: note: in expansion of macro 'TERN_'
     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
     ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:228:57: error: 'PAUSE_RESPONSE_RESUME_PRINT' was not declared in this scope
     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
                                                         ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:443:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:192:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:191:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:228:5: note: in expansion of macro 'TERN_'
     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
     ^~~~~
Compiling .pio\build\mks_robin_nano35\src\src\lcd\extui\lib\mks_ui\draw_hotend_offset_settings.cpp.o
[... cut ...]
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_dialog.cpp:228:5: note: in expansion of macro 'TERN_'
     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
     ^~~~~
Compiling .pio\build\mks_robin_nano35\src\src\lcd\extui\lib\mks_ui\draw_jerk_settings.cpp.o
*** [.pio\build\mks_robin_nano35\src\src\lcd\extui\lib\mks_ui\draw_dialog.cpp.o] Error 1`

zonalimitatore, 2021-05-27:
No one?