makerbase-mks/MKS-Robin-Nano-V2.X #19: Bootloader with USB support

lecler-i, 2020-11-25:
Hello,

What I understand is that STM32F103VET6 does not come with usb bootloader (dfu) out of the box, and flashing is only available via serial ports (in addition of sdcard) ?

I would like to flash dfu bootloader so I am able to flash via usb and use serial over usb.
I'm very new to soc world, is it safe to follow a procedure like https://www.youtube.com/watch?v=Myon8H111PQ ?

I found https://github.com/Serasidis/STM32_HID_Bootloader, would it be Ok to apply such bootloader on Mks Robin Nano v2 ? Or you have a custom bootloader already ?

thomas

mks-viva, 2020-11-26:
Haha! Nano V2 bootloader file : https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/tree/master/Bootloader

marchaos, 2020-12-10:
Hey @lecler-i - any luck getting this working?

mks-viva, 2020-12-30:
Hi! If have not response, i will close this issues temporarily.

closed 2020-12-30 by mks-viva

marchaos, 2020-12-30:
Can someone at MKS advice as to how we can use the bootloader above? Perhaps some instructions as to how to connect an ST-LINK to the Robin Nano?

mks-viva, 2020-12-31:
@marchaos 
Thank you for your feedback, we will test and improve when use ST-LINK upload firmware.