makerbase-mks/MKS-Robin-Nano-V2.X #20: My ts35 is broken i want to replace it,which model do i need to buy

haxiny, 2020-12-01:
My ts35 is broken,i want to replace it. Is it the ili9488? Or other.Thank you!

mks-viva, 2020-12-01:
It is ST7796S. I suggest you use the firmware to test it first. it is doesn’t working, replacing LCD.

haxiny, 2020-12-01:
Thank you,i will test it.i will buy the same tft------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;mks-viva&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年12月1日(星期二) 下午4:27
收件人:&nbsp;&quot;makerbase-mks/MKS-Robin-Nano-V2.X&quot;<MKS-Robin-Nano-V2.X@noreply.github.com&gt;;
抄送:&nbsp;&quot;haxiny&quot;<6046246@qq.com&gt;;&quot;Author&quot;<author@noreply.github.com&gt;;
主题:&nbsp;Re: [makerbase-mks/MKS-Robin-Nano-V2.X] My ts35 is broken i want to replace it,which model do i need to buy (#20)

haxiny, 2020-12-01:
Thank you! I will buy the same model!

closed 2020-12-01 by haxiny