makerbase-mks/MKS-Robin-Nano-V2.X #21: MKS FES on ROBIN NANO V2 + TS35 = compilation error

Oranginam, 2020-12-02:
-MKS Robin Nano V2
-MKS TMC2209 (x5) in uart mode
-MKS TS35 (with encoder
-MKS FES
-Firmware Marlin-bugfix2.0.x-MKS-2.1.1

I have followed your wiki :

In configuration.h
-Enable #define FILAMENT_RUNOUT_SENSOR
-Enable #define NOZZLE_PARK_FEATURE

In pins_MKS_ROBIN_NANO_V2.h
-Add #define FIL_RUNOUT_PIN PA4 //You can change to other pin

In confiuration_adv.h
-Enable #define ADVANCED_PAUSE_FEATURE

And get compilation error:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:836:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."

In confiuration_adv.h there is this commented line "Requires an LCD display".

Please fix this problem, thx

mks-viva, 2020-12-02:
Hi! Thanks for your feedback, I will test it and fix it.

demeiji, 2020-12-09:
> -MKS Robin Nano V2
> -MKS TMC2209 (x5) in uart mode
> -MKS TS35 (with encoder
> -MKS FES
> -Firmware Marlin-bugfix2.0.x-MKS-2.1.1
>
> I have followed your wiki :
>
> In configuration.h
> -Enable #define [...]

Hi
this error is being commented in can't use filament detector with ts35
regards

Oranginam, 2020-12-09:
Hi dont understand. In MKS WIKI :

MKS Robin Nano V2 Firmware setting

In configuration.h

Enable #define FILAMENT_RUNOUT_SENSOR

Enable #define NOZZLE_PARK_FEATURE

In pins_MKS_ROBIN_NANO_V2.h

Add #define FIL_RUNOUT_PIN PA4 //You can change to other pin

In confiuration_adv.h

Enable #define ADVANCED_PAUSE_FEATURE

How to use

Work with MKS TS35

Usage method

If the material is interrupted or out of material during printing, it will be triggered and enter the pause interface

You need reload supplies and resume printing

misiek271, 2020-12-09:
filament sensor with ts35 works bad MKS try to fix it but for now its still broke.

mks-viva, 2020-12-10:
I has update it: https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/MKS_FES

misiek271, 2020-12-10:
Nice. But what about problem unloading filament after reasume print? As on my video? Undefined runout script wasnt help. Its still the same behavior as on my video.

Oranginam, 2020-12-10:
nice ?
I bought this board (with TS35, 5 x tmc2209 + FES + PWC v3) after carefully reading the wiki. I want a real fix and use M600 not a wiki fix ....

mks-viva, 2020-12-11:
@Oranginam 
Hi! Wiki is just a tutorial, we will continue to improve it

misiek271, 2020-12-18:
Any news About Bad working filament script and unloading filament after reasume?

mks-viva, 2020-12-19:
> Any news About Bad working filament script and unloading filament after reasume?

https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/issues/18#issuecomment-748178240

UrknallBigBang, 2020-12-19:
Try this if you use MKS UI and PWC 
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/140#issuecomment-743805236

don't define 
// #define FILAMENT_RUNOUT_SENSOR.
//#define ADVANCED_PAUSE_FEATURE
//#define NOZZLE_PARK_FEATURE
//#define PSU_CONTROL

the only thing you have to do is written in the link

misiek271, 2020-12-20:
I use ts35 with ribin nano v2 board and i dont define it and still resume print dont work coz when i try to reasume extruder unload filament and then start printing

UrknallBigBang, 2020-12-20:
My hardware config is
MKS Robin nano V2 with TS35
MKS PWC 3.0
MKS TMC2209 2x
MKS Servo42b 2x

Don't follow the Wiki it doesn't work for MKS UI only for Marlin UI. The only thing you have to define is TMC2209 or type of your own. Don't define FIL_RUNOUT_PIN PA4 that's only nessesary if you use Marlin UI.

You don't need to define PSU_CONTROL you have to define in pins_MKS_ROBIN_NANO_V2.h (for Robin Nano V2) like this
(must be the same in pins_MKS_ROBIN_NANO.h defined for Robin Nano V1.X)

That is my pin config for PWC and Filament Runout

#if HAS_TFT_LVGL_UI
  #define MKSPWC
  #ifdef MKSPWC

    #define SUICIDE_PIN                       PB2   // Enable MKSPWC SUICIDE PIN
    #define SUICIDE_PIN_INVERTING             false // Enable MKSPWC PIN STATE
    #define KILL_PIN                          PA2   // Enable MKSPWC DET PIN
    #define KILL_PIN_STATE                    HIGH  // Enable MKSPWC PIN STATE
  #endif

  #define MT_DET_1_PIN                        PA4   // LVGL UI FILAMENT RUNOUT1 PIN
  #define MT_DET_2_PIN                        PE6   // LVGL UI FILAMENT RUNOUT2 PIN
  #define MT_DET_PIN_INVERTING                false // LVGL UI filament RUNOUT PIN STATE
 
  #define WIFI_IO0_PIN                        PC13  // MKS ESP WIFI IO0 PIN
  #define WIFI_IO1_PIN       			            PC7   // MKS ESP WIFI IO1 PIN
  #define WIFI_RESET_PIN				              PE9   // MKS ESP WIFI RESET PIN

  #if ENABLED(MKS_TEST)
    #define MKS_TEST_POWER_LOSS_PIN           PA2   // PW_DET
    #define MKS_TEST_PS_ON_PIN                PB2   // PW_OFF
  #endif
#else
  #define POWER_LOSS_PIN                      PA2   // PW_DET
  #define PS_ON_PIN                           PB2   // PW_OFF
  #define FIL_RUNOUT_PIN                      PA4
  #define FIL_RUNOUT2_PIN                     PE6
#endif

And you have to hold the button from MKS PWC when you do the update. When update is done then switch on the printer and hold button until MKS Screen appeares. When you hold too long then Error Printer hold appears. Do it once more an let the button go when MKS Screen appeares. All done have fun.

misiek271, 2020-12-20:
[
[Configuration_adv.txt](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5721087/Configuration_adv.txt)
[Configuration.txt](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5721088/Configuration.txt)
[pins_MKS_ROBIN_NANO_V2.txt](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5721089/pins_MKS_ROBIN_NANO_V2.txt)

](url)

i send my files. everyting is like you sad exept kill pin coz i dont use it. and still. reasume after power loss works good. but reasume after normal pause button works bad . after reasume extruder unload filament.

Oranginam, 2020-12-20:
Hello

I opened this ticket for the FES, not for the PWC.
The idea being to be able to use the M600 (ADVANCED_PAUSE_FEATURE) as indicated initially.

Thx

misiek271, 2020-12-21:
My problem has got nothing with pwc. Its fes problem. Fes and pause button.. Acording to behavior uses the same function coz fes dont work. And reasume button dont work. They unl9ad filament before after click reasume

mks-viva, 2020-12-21:
@UrknallBigBang Thank you

Oranginam, 2021-03-08:
Hu?
Why Closer ?

M600 work or it's a lie... ?

mks-viva, 2021-03-08:
@Oranginam 
https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/MKS_FES
[image]

misiek271, 2021-03-08:
Not working. They never fix it. Lvl GUI its not Worth it. Only color marlin ui works good

artdezigner, 2021-11-30:
> @Oranginam https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/MKS_FES [image]

ЭТО НЕ РАБОТАЕТ!!!