makerbase-mks/MKS-Robin-Nano-V2.X #24: Robin nano v2 + ts35 mks ui. Reasume after pause dont work

misiek271, 2020-12-21:
Hello
We have probkem with reasume after pause or filament runout sensor coz they behavior the same. 
When click on pause button printer stops printing and goes to position that i set in ts35. The same goes with filament sensor.
But. After click reasume printer goes to latest position(like it should)  and then start unload 
filament(this shoud never happen) and then start print.

Something is wrong with rasume script. I try every combination of define undefine  filamemt sensor etc. Try the latest marlin from your github. Problem is the same. 
[Video](https://drive.google.com/file/d/1AOFw5nIoWU7n_m7NQ1LnA39rWS3ZuwuH/view?usp=drivesdk)

mks-viva, 2020-12-30:
Hi! I have reported this problem to the engineer, thank you for your feedback.

misiek271, 2021-01-21:
Hi. 
did YOu fix this problem or You just leave it? any news?

mks-viva, 2021-01-25:
@misiek271 I forgot to tell you,  the latest version firmware has fixed this bug.

misiek271, 2021-01-27:
haha. still not working. now after reasume filament does not retract(so its one step forward)  but extruder not printing. make only linear advence moves to simplify xyz axis work great but E0 not make any movment after reasume. Only little LA movment. so behavior like on end of my videos.