makerbase-mks/MKS-Robin-Nano-V2.X #25: Another Micro Sd Connector

SimoneCarnio, 2020-12-29:
Can I connect a external micro sd connector to the J4 connector ?  Looking at the schematics seems directly connected to the sd slot. 

[image]

[image]

mks-viva, 2020-12-30:
Hi! Yes you can. But you need to pay attention to the location of J4
[image]

SimoneCarnio, 2020-12-30:
Ya sorry, I take the wrong screenshot, my fault (that was for another project :D)

closed 2020-12-30 by SimoneCarnio