makerbase-mks/MKS-Robin-Nano-V2.X #26: Servo and Probe not declared in tft_lvgl_configuration

LeonardoLH, 2021-01-24:
Hi! I recently bought the MKS Robin Nano 2.0 board and an MKS TS35 2.0 display. When downloading the firmware available here on github (Mks-Robin-Nano-Marlin2.0-Firmware) and trying to configure it for my hardware I had the following error:

```
In file included from Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/boards.h:24:0,
                 from Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp:22:
Marlin\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp: In function 'void tft_lvgl_init()':
Marlin\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp:205:21: error: 'servo_init' was not declared in this scope
   TERN_(HAS_SERVOS, servo_init());
                     ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:419:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:194:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:193:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:192:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp:205:3: note: in expansion of macro 'TERN_'
   TERN_(HAS_SERVOS, servo_init());
   ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp:205:21: note: suggested alternative: 'lv_init'
   TERN_(HAS_SERVOS, servo_init());
                     ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:419:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:194:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:193:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:192:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp:205:3: note: in expansion of macro 'TERN_'
   TERN_(HAS_SERVOS, servo_init());
   ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp:206:28: error: 'probe' was not declared in this scope
   TERN_(HAS_Z_SERVO_PROBE, probe.servo_probe_init());
                            ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:419:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:194:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:193:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin\src\lcd\extui\lib\mks_ui\../../../../inc/../core/macros.h:192:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
Marlin\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp:206:3: note: in expansion of macro 'TERN_'
   TERN_(HAS_Z_SERVO_PROBE, probe.servo_probe_init());
   ^~~~~
*** [.pio\build\mks_robin_nano35\src\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp.o] Error 1
```
And when I try to configure the MKSTool website, the display icons are buggy as follows:

[image]

My hadrware are the Ender 3 with the followings mods:
Nozzle thermistor type 5
3DTouch
MKS FES
TMC 2209 2.0 on UART Mode and Sensorless Homing
MKS WiFi module

mks-viva, 2021-01-25:
Hi! Send me your configuration file, or the entire firmware?

LeonardoLH, 2021-01-25:
Ok!
Here my firmware
[firmware.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/files/5866502/firmware.zip)

mks-viva, 2021-01-26:
Ok! We will test it.

LeonardoLH, 2021-02-02:
Ok, I'm waiting