makerbase-mks/MKS-Robin-Nano-V2.X #30: est il possible de configurer deux axe Z avez la MSK ROBIN NANO V2

vosphil, 2021-03-04:
Bonjour, je possède une imprimante avec deux moteur pour l'axe Z, est t'il possible de configurer la carte pour alimenter les deux moteur?
merci

leofaure, 2021-03-06:
Saludos!!
#define NUM_Z_STEPPER_DRIVERS 2

esto convierte el E1 en motor Z2.

vosphil, 2021-03-08:
ok, merci pour la réponse

adadrag, 2021-03-29:
Hi There,
I have the same issue.
I defined in configuration_adv.h :
#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many

But the step motor connected to the E1 is spinning faster then the one connected on Z.
There is other setting to fix this?
Thanks in advance

leofaure, 2021-03-29:
Hola, tube ese mismo problema me he dado cuenta que estaban mal conectados los jumpers del driver.
Aqui te dejo un enlace de Edward Braiman https://www.youtube.com/watch?v=2Hy9frwsFD4, donde explica hacerlo como con una MKS Gen, la configuración es similar.A mi me funciona ahora con mis tmc 2209. si necesitas mi código contactame y te lo haré llegar de alguna manera.
buena suerte!

adadrag, 2021-03-29:
@leofaure Thanks for your comments.
The video is for Marlin v1.x.x and is not applicable for Marlin v2.0.6
Can you let me know how you setup the jumpers please?
Do you use Marlin v2.x.x?