makerbase-mks/MKS-Robin-Nano-V2.X #32: Sapphire Pro configuration problem

Micropirate, 2021-05-12:
Hi, I'm going around in my head to solve this problem. I switched the board of my 3D nano v1.2 printer to a robin nano v2.0 with TS35 display. I've changed my printer's firmware several times, only I'm not getting it to work properly. Her autohome should be in the middle, and she goes to the corner, she's a coreyx. I can't do it. I changed it to marlin's Two Trees Sapphire Pro firmware, and compiled the information that was needed. To no avail. Somebody help me.

themacboy, 2021-05-12:
What probe do you use?

Missatge de Micropirate ***@***.***> del dia dc., 12 de maig
2021 a les 20:45:

Micropirate, 2021-05-12:
@themacboy I don't use anything, it's source, I just changed the board and the screen.

themacboy, 2021-05-12:
If you don't use any probe,  only endpapers then you machine is working
correctly, the home is done in the corner you set when you compile marlin.

On the center of bed is usually done when you use a probe.

Otherwise you can define it in the main configs.

I think you must take a few classes of marlin configuration. Youtube is
full of them.

Micropirate, 2021-05-12:
@themacboy ok thanks, I will then look for how to install a BLTouch on my sapphire pro with robin nano v2.0

themacboy, 2021-05-12:
In sapphire pro by default home is fine in the min X and in themax Y corner.

Micropirate, 2021-05-12:
@themacboy Thanks for help!

themacboy, 2021-05-12:
Check at mks youtube, there are a few videos about configuring bltouch