makerbase-mks/MKS-Robin-Nano-V2.X #33: ESP3D work only in MKS UI?

DvoraNoob, 2021-05-19:
I would like to know, if ESP3D work only in MKS UI, because I have mks wifi, but i cant use MKS UI.

mks-viva, 2021-05-19:
ESP3D firmware can be work in web control, you can use it in 12864 stype

DvoraNoob, 2021-05-19:
> ESP3D firmware can be work in web control, you can use it in 12864 stype

I want use wifi in Marlin Color Ui, but doesn't load the firmware to board

themacboy, 2021-05-19:
You can load Esp3d firmware in mks ui, and later restarts in marlin color
ui.

Other wise you need a flasher to update mks WiFi card

DvoraNoob, 2021-05-19:
> You can load Esp3d firmware in mks ui, and later restarts in marlin color ui. Other wise you need a flasher to update mks WiFi card El mié., 19 may. 2021 3:17, DvoraNoob ***@***.***> escribió:
> […](#)

Doesn't work, the wifi SSID not show, so de module is off

themacboy, 2021-05-19:
Under MKS UI it works. I update it a lot of times. You simply need to
naming the file correctly, restart in MKS UI mode, and wait for update.
Later re-activate the Marlin Color UI and all works fine.

My 3 printers works the same.

Missatge de DvoraNoob ***@***.***> del dia dc., 19 de maig
2021 a les 23:40:

DvoraNoob, 2021-05-19:
I compile the ESP3D 3.0 edited, after I rename de bin file to 'MksWifi.bin', after 
I loaded in the 3D printer using the MKS UI, after this work normally in MKS UI, but I'm losing steps in MKS UI, after I tested in MKS UI I loaded the Marlin Color UI, so the Wifi of the printer doens't work more.

I have acess to the webUI, but any movement happens

DvoraNoob, 2021-05-20:
> The Esp3d Webui is independent to board ui. ‎Tente se conectar diretamente com seu celular para Esp3d ssdi, c e, em seguida, acessar a Interface web B nos endereços ip 192.168.0.1. Em seguida, configure todas as configurações do‎ WiFi El jue., 20 may. 2021 1:37, DvoraNoob ***@***.***> escribió:
> […](#)

Same problem, the WebUi load, but nothing happens when I click move, end the SD stay with infinite loading

DvoraNoob, 2021-05-20:
> Have you select the correct serial port in marlin?

My serial:
[image]

DvoraNoob, 2021-05-20:
> > Have you select the correct serial port in marlin?
> 
> My serial port:
> [image]

DvoraNoob, 2021-05-20:
In esp3d  configuration.h a I edited only the COMMUNICATION_PROTOCOL to MKS_SERIAL and the esp3d settings to mks wifi module

[image]
[image]

themacboy, 2021-05-20:
> 
> 
> > > Have you select the correct serial port in marlin?
> > 
> > 
> > My serial port:
> > [image]

[image]
  
  As I read in some palce the serial_port_2 must be 1 !

DvoraNoob, 2021-05-20:
> > > > Have you select the correct serial port in marlin?
> > > 
> > > 
> > > My serial port:
> > > [image]
> 
> [image]
> 
> As I read in some palce the serial_port_2 must be 1 !
> 
> I don0t understand the reason you set port_1 to 3

I only follow the instruction of Robin Nano V2

themacboy, 2021-05-20:
> 
> 
> In esp3d configuration.h a I edited only the COMMUNICATION_PROTOCOL to MKS_SERIAL and the esp3d settings to mks wifi module
> 
> [image]
> [image]

I haven't changed any thing in the ESP3D configs.

DvoraNoob, 2021-05-20:
> > In esp3d configuration.h a I edited only the COMMUNICATION_PROTOCOL to MKS_SERIAL and the esp3d settings to mks wifi module
> > [image]
> > [image]
> 
> I haven't changed any thing in the ESP3D configs.

[image]

themacboy, 2021-05-20:
> 
> 
> > > In esp3d configuration.h a I edited only the COMMUNICATION_PROTOCOL to MKS_SERIAL and the esp3d settings to mks wifi module
> > > [image]
> > > [image]
> > 
> > 
> > I haven't changed any thing in the ESP3D configs.
> 
> [image]

I use raw_serial in port_2 number 1. And it is wroking for me.

DvoraNoob, 2021-05-20:
with serial_port_2 1 same problem, now I'll try with Raw_Serial

themacboy, 2021-05-21:
It must work in mks_serial too.

Try other port_2 number.

DvoraNoob, 2021-05-21:
I tried 1 e 3, negative numbers doesn't compile, I'll try others number

themacboy, 2021-05-21:
> 
> 
> I tried 1 e 3, negative numbers doesn't compile, I'll try others numbers

In robin v.3 you have defined serial prots for wifi in pins file:

[image]

But unfortunatelly in v1.2 and v.2 it is not defined.

Better ask deirectlly to makerbase support service ( @mks-viva  )

DvoraNoob, 2021-05-21:
I think MKS has changed something that makes it not work, because these settings are only defined there, or it is not really compatible

themacboy, 2021-05-21:
Have you tryed serial _port_2 0?

DvoraNoob, 2021-05-21:
> […](#)

yes, there was a compilation error

closed 2021-08-09 by DvoraNoob