makerbase-mks/MKS-Robin-Nano-V2.X #36: Freezing problem with screen and Board

TechnicalWorld2016, 2021-08-04:
Hi 
I have been using MKS Robin Nano V2 and TS35 for all the printers.
The Printers are configured well and are able to get positive results. The printers were installed at clients locations and they using them.

There was some issues freezing with the printers/ board and screen.
the prints are stopping automatically while printing the screen will be freezed and there will not be any error message or any moments in the printer. The hot-end will be ideal and parked at the parking location.  When we see it the screen is not responding to the touch. when we turn off the machine and turn it on again the screen does not show resume option some times and it shows resume option some times. 

If we give resume the print continues and tries to print in air around 2-3 mm. Some there will not any option for resume also.

I have checked in community and I am sure that there is no electrical interference among the electronics or screen. Even the screen is grounded properly where there is a ground option provided to the screen.

But this issue is getting with some machines repeatedly at client locations.

I am not sure where is error. All the configurations are good. Connections are proper. and the problem is not in all the prints.

So, someone please help me with this regard. And give me some suggestions or solutions which can help solving my issue.

Thank you

mks-viva, 2021-09-13:
Can you update the latest version firmware and test it?

tiagopizzi, 2022-08-05:
Hi
I have the same problem.

You fixed this?

How did you solved it?